rusefi/hellen125honda-issues #3: 4 wire DENSO wbo

rusefillc, 2022-12-19:
We shall provision for OEM 4 wire middle band sensor control via https://github.com/mck1117/denso-wideband

[image]

rusefillc, 2022-12-19:
@Qwerty-OFF should https://github.com/mck1117/denso-wideband become a hellen module probably?

rusefillc, 2023-01-09:
EU PRA
[image]

US PRB
[image]

Qwerty-OFF, 2023-01-10:
Если мы не будем часто этим пользоваться давай соберу из компонентов, не будем делать лишний модуль.

rusefillc, 2023-01-10:
Ок, давай пока так!

rusefillc, 2023-01-12:
[image]

https://github.com/rusefi/rusefi_documentation/blob/master/OEM-Docs/Honda/2005-rsx-s-1.png

yes A1 heater with VND10

A6 is "positive" and A16 is "negative" signal wires for "4 wire denso" module

A16 not ground

We only care about primary sensor for now so E2 and E4 are just vias.

Qwerty-OFF, 2023-01-18:
Done

closed 2023-01-18 by Qwerty-OFF