rusefi/hellen125honda-issues #7: alternator

rusefillc, 2023-01-09:
[image]

reminds me of Hellen72 NB2 miata - that's the only other board where we had alternator control success

https://github.com/rusefi/rusefi_documentation/blob/master/Hardware/Hellen/hellen72-d-schematic.pdf

rusefillc, 2023-01-10:
нам НАДО: В18 ALTC - это управление генератором с мозга МИНУСОМ. ВОобще сделано для экономии топлива на круизах, при подаче минуса генератор ограничивается до 12.5-13в.

просто via: В13 ALTFR - Это выход нагрузки.  он стандартный на многих генераторах всех машин, передает по своему протоколу значение нагрузки на генер. точно не знаю что именно но пишут 0-5v в зависимости от нагрузки

В10 - ALTL - тут интересно. вообще это просто лампа. и там на заглушенную приходит 0, на заведенную 12в. в хондах старых на приборке на лампе генера 12в всегда висит, и тут уже может от мозга\модели авто зависесть, есть ли такой выход и нужен ли он в  мозгах.

rusefillc, 2023-01-10:
[image]

Qwerty-OFF, 2023-01-10:
[image]
[image]
[image]

rusefillc, 2023-01-10:
B10 might be an INPUT of alternator error light which goes to dash?!

it's confusing and probably not important, let's DNP corresponding resistor

rusefillc, 2023-01-10:
@Qwerty-OFF let's make B10 just a via. no need for B10 output, it's really an input

Qwerty-OFF, 2023-01-11:
[image]

closed 2023-01-11 by Qwerty-OFF