rusefi/hellen125honda-issues #16: hall into oscilloscope

rusefillc, 2023-02-01:
let's hook crank and both cams into ADC3 for oscilloscope

we seem to have spare digital inputs to move what is currently digital

rusefillc, 2023-02-01:
just a bit related to https://github.com/rusefi/hellen125honda-issues/issues/16

related to https://github.com/rusefi/hellen154hyundai/issues/82

Qwerty-OFF, 2023-02-01:
[image]

closed 2023-02-01 by Qwerty-OFF

rusefillc, 2023-02-01:
@mck1117 this makes me wonder if we want to combine ADC and trigger input in one pin?

Well, that starts to look like ADC-based trigger input like in hellen81

Unless it's all totally pointless for Hall anyway :)

mck1117, 2023-02-02:
> this makes me wonder if we want to combine ADC and trigger input in one pin?

wait, are we routing different things here? what is the point?

which thing do we connect to ADC, the raw pin or after the digital input buffer?

> Unless it's all totally pointless for Hall anyway :)

It is pointless for hall, because it's a digital signal.