rusefi/hellen125honda-issues #19: rev A sleep consumes too much

rusefillc, 2023-02-11:
0.080A running

```
canRxAdd(0x570)

function onCanRx(bus, id, dlc, data)
	print('got CAN id=' ..id ..' dlc=' ..dlc)
	mcu_standby()
end

function onTick()
end

```

```
	txCan(1, 0x570, 0, {0} )
function onTick()
end

```

rusefillc, 2023-02-14:
R630 removal is a 0.013A win, removal of three LEDs is about 0.002A

baseline ACTIVE (now with EN logic note while LEDs on the left)

[image]

baseline SLEEP (note VTS while LED still ON)
[image]

first step: removal of two left blue LEDs from power supply 0.001A improvement
[image]

second step R630 removal
[image]

third step 3.3v led removal from mm
[image]

Qwerty-OFF, 2023-02-19:
@rusefillc check 5v in proto area in sleep mode

rusefillc, 2023-02-27:
@qwerty-off новая плата. выпаял R630

rusefillc, 2023-02-27:
[image]

[image]

rusefillc, 2023-04-02:
to be continued on any mm-0.3 board

closed 2023-04-02 by rusefillc