rusefi/hellen125honda-issues #23: CAN resistor should it be software-configurable?

rusefillc, 2023-02-15:
I assume we have 120R somewhere for CAN? can we have it configurable for cases of external CAN already having resistors?

rusefillc, 2023-02-16:
@mck1117 @eldominio what do you think are there use-cases and reasons to have configurable 120R?

Qwerty-OFF, 2023-02-19:
Now it is always soldered

The easiest way to configure:
[image]
Or specify DNP in BOM

rusefillc, 2023-02-19:
Jumper vibration bad, manual soldering too complex for our customer base.

We definitely want to learn software-defined CAN termination this board seems like a great place to try it

Qwerty-OFF, 2023-02-24:
[image]

closed 2023-02-28 by Qwerty-OFF