rusefi/hellen125honda-issues #25: tach does not work

rusefillc, 2023-02-17:
Is it high side or low size?

At the moment we have low side

[image]

rusefillc, 2023-02-19:
Next step to try high-side control

Qwerty-OFF, 2023-02-24:
Делаю high-side в rev. b?

rusefillc, 2023-02-24:
пока нет, попробую во вторник-среду всё-таки подствердить

rusefillc, 2023-03-03:
[image]

[image]

[image]

[image]

rusefillc, 2023-03-04:
P18 did the trick!

since we use P18 for something do we have digital pins to have second pushpull available as aux?

Qwerty-OFF, 2023-03-04:
[image]
PP2 (OUT_IO10) is tacho now. P18 not changed

rusefillc, 2023-03-04:
let's please remove todo

[image]

Qwerty-OFF, 2023-03-11:
Done

closed 2023-03-11 by Qwerty-OFF