rusefi/hellen125honda-issues #27: rev A does not read K-line bytes

rusefillc, 2023-02-17:
[image]

[image]

[image]

rusefillc, 2023-02-18:
I've soldered wires to RX and TX pins of U440 and oscilloscope was showing nothing

I've poked DMM into pin2 and saw 5v

I've poked DMM into pin3 and saw 12v

I've then shorted pin3 to pin4 and this board is no longer blinking with stm32 I assume this stm32 is now dead after 12v on UART TX? i have more boards please double check LIN module schematics @Qwerty-OFF @andreika-git

andreika-git, 2023-02-18:
Maybe we don't set efiSetPadMode() for UART2?

rusefillc, 2023-02-18:
@andreika-git says to remove R440

rusefillc, 2023-02-18:
https://www.nxp.com/docs/en/data-sheet/TJA1021.pdf

[image]

rusefillc, 2023-02-18:
got hardware progress, software still unhappy

https://github.com/andreika-git/hellen-one/issues/261

[image]

rusefillc, 2023-02-19:
Fun experiment

discovery board happy at 9600
[image]

[image]

rusefillc, 2023-02-19:
happy 14400

[image]
[image]

rusefillc, 2023-02-19:
unhappy 10400

[image]
[image]

rusefillc, 2023-02-21:
https://rusefi.com/forum/viewtopic.php?p=47201#p47201

rusefillc, 2023-02-24:
https://github.com/rusefi/rusefi/pull/5125

[image]

closed 2023-02-24 by rusefillc