rusefi/hellen125honda-issues #28: SD card status unclear

rusefillc, 2023-02-18:
SD card does not show as MSD

could be https://github.com/rusefi/rusefi/issues/5097

rusefillc, 2023-02-19:
* something is not great with board PDF - search for "SD_CS" does not find me SD_CS text
when I search for MOSI at least I see SD_MOSI but I only see it once, how do I know where is SD card connected to?!

[image]

Qwerty-OFF, 2023-02-19:
Use SPI1 with PB6 CS

rusefillc, 2023-02-19:
@Qwerty-OFF where can I see this on the PDF?

rusefillc, 2023-02-19:
just changed to SPI1 and still does not work
```
2023-02-19_13_49_52_243: EngineState: pin PB3: SPI clock
2023-02-19_13_49_52_247: EngineState: pin PB4: SPI master in 
2023-02-19_13_49_52_248: EngineState: pin PB5: SPI master out
2023-02-19_13_49_52_249: EngineState: pin PB6: SD CS
```

rusefillc, 2023-02-19:
PDF search issue starts at hellen-one/modules/mega-mcu144/0.2/mega-mcu144-schematic.pdf

if I search for "_mosi" it gives me ACC_MOSE spi1_mosi  spi2_mosi spi3_mosi but nothing about SD

rusefillc, 2023-02-19:
Good news SD card lines are going at least somewhere :)

[image]

rusefillc, 2023-02-26:
[image]

dron0gus, 2023-02-26:
~~"I'm not going to have per-board board.c file."~~
All f4 boards use some random board.c file from STM F4 Discovery board https://github.com/rusefi/rusefi/blob/d1e35213c66ddfc15cd502b980102abe0b2af917/firmware/hw_layer/ports/stm32/stm32f4/hw_ports.mk#L11
`__early_init()` is also used from this file. It calls `stm32_gpio_init()` that is again Discovery-board-specific.
This function set PA5, PA6, PA7 for SPI1 function.
Does any other code reset SPI1 function for these pins before setting PB3, PB4, PB5?

rusefillc, 2023-02-26:
@dron0gus same nothing :(

[image]

closed 2023-02-27 by rusefillc