rusefi/hellen125honda-issues #43: How to make Coolant temp gauge work?

ElDominio, 2024-02-23:
From what I understand, this:
[image]

Makes coolant temp gauge work. But I have two clients in the field that report no coolant gauge, an EP3 Civic and a 2002 RSX. Any other magic? BTW this is ancient firmware (may 11 2023) and I need to log in to update

ElDominio, 2024-03-30:
[image]

[image]

[image]

2003 Acura RSX

rusefillc, 2024-03-31:
@ElDominio I see you already have verbose K enabled

next step is to use rusEFI console to see what it says

ElDominio, 2024-03-31:
`te: kline doSend
2024-03-31_08_58_30_425: EngineState: kline: got count 0x07
2024-03-31_08_58_30_426: EngineState: kline: got 0x02
2024-03-31_08_58_30_426: EngineState: kline: got 0x00
2024-03-31_08_58_30_426: EngineState: kline: got 0x00
2024-03-31_08_58_30_426: EngineState: kline: got 0x1F
2024-03-31_08_58_30_426: EngineState: kline: got 0x00
2024-03-31_08_58_30_426: EngineState: kline: got 0x00
2024-03-31_08_58_30_426: EngineState: kline: got 0x44
2024-03-31_08_58_30_426: EngineState: happy CRC 0x44
2024-03-31_08_58_30_426: EngineState: kline: got count 0x05
2024-03-31_08_58_30_430: EngineState: kline: got 0x00
2024-03-31_08_58_30_433: EngineState: kline: got 0xD0
2024-03-31_08_58_30_436: EngineState: kline: got 0x04
2024-03-31_08_58_30_436: EngineState: kline: got 0x00
2024-03-31_08_58_30_439: EngineState: kline: got 0x35
2024-03-31_08_58_30_439: EngineState: happy CRC 0x35
2024-03-31_08_58_30_439: EngineState: honda status packet with 0xD0 0x04 state 0
2024-03-31_08_58_30_439: EngineState: kline doSend
2024-03-31_08_58_30_439: EngineState: kline: got count 0x01
2024-03-31_08_58_30_439: EngineState: kline: got 0x02
2024-03-31_08_58_30_439: EngineState: kline: got count 0x01
2024-03-31_08_58_30_439: EngineState: kline: got 0x00
2024-03-31_08_58_30_440: EngineState: kline: got count 0x02
2024-03-31_08_58_30_440: EngineState: kline: got 0x00
2024-03-31_08_58_30_440: EngineState: kline: got 0x44
2024-03-31_08_58_30_440: EngineState: kline: got count 0x04
2024-03-31_08_58_30_440: EngineState: kline: got 0x01
2024-03-31_08_58_30_440: EngineState: kline: got 0x00
2024-03-31_08_58_30_440: EngineState: kline: got 0x00
2024-03-31_08_58_30_440: EngineState: kline: got 0x8C
2024-03-31_08_58_30_440: EngineState: happy CRC 0x8C
2024-03-31_08_58_30_441: EngineState: BCM request 0x01
2024-03-31_08_58_30_441: EngineState: kline doSend
2024-03-31_08_58_30_441: EngineState: kline: got count 0x07
2024-03-31_08_58_30_441: EngineState: kline: got 0x02
2024-03-31_08_58_30_441: EngineState: kline: got 0x00
2024-03-31_08_58_30_441: EngineState: kline: got 0x00
2024-03-31_08_58_30_441: EngineState: kline: got 0x1F
2024-03-31_08_58_30_441: EngineState: kline: got 0x00
2024-03-31_08_58_30_442: EngineState: kline: got 0x00
2024-03-31_08_58_30_442: EngineState: kline: got 0x44
2024-03-31_08_58_30_442: EngineState: happy CRC 0x44
2024-03-31_08_58_30_442: EngineState: kline: got count 0x05
2024-03-31_08_58_30_442: EngineState: kline: got 0x00
2024-03-31_08_58_30_442: EngineState: kline: got 0xD0
2024-03-31_08_58_30_442: EngineState: kline: got 0x04
2024-03-31_08_58_30_449: EngineState: kline: got 0x00
2024-03-31_08_58_30_449: EngineState: kline: got 0x35
2024-03-31_08_58_30_450: EngineState: happy CRC 0x35
2024-03-31_08_58_30_450: EngineState: honda status packet with 0xD0 0x04 state 0
2024-03-31_08_58_30_450: EngineState: kline doSend
2024-03-31_08_58_30_450: EngineState: kline: got count 0x01
2024-03-31_08_58_30_450: EngineState: kline: got 0x02
2024-03-31_08_58_30_450: EngineState: kline: got count 0x01
2024-03-31_08_58_30_450: EngineState: kline: got 0x04
2024-03-31_08_58_30_451: EngineState: kline: got count 0x02
2024-03-31_08_58_30_451: EngineState: kline: got 0x00
2024-03-31_08_58_30_451: EngineState: kline: got 0x44
2024-03-31_08_58_30_489: EngineState: kline: got count 0x04
2024-03-31_08_58_30_489: EngineState: kline: got 0x01
2024-03-31_08_58_30_489: EngineState: kline: got 0x00
2024-03-31_08_58_30_489: EngineState: kline: got 0x00
2024-03-31_08_58_30_490: EngineState: kline: got 0x8C
2024-03-31_08_58_30_491: EngineState: happy CRC 0x8C
2024-03-31_08_58_30_493: EngineState: BCM request 0x01
2024-03-31_08_58_30_494: EngineState: kline doSend
2024-03-31_08_58_30_501: EngineState: kline: got count 0x07
2024-03-31_08_58_30_503: EngineState: kline: got 0x02
2024-03-31_08_58_30_504: EngineState: kline: got 0x00
[... 520 lines cut ...]
2024-03-31_08_58_31_089: EngineState: kline: got count 0x08
2024-03-31_08_58_31_089: EngineState: kline: got 0x8C
2024-03-31_08_58_31_089: EngineState: kline: got 0x02
2024-03-31_08_58_31_090: EngineState: kline: got 0x00
2024-03-31_08_58_31_090: EngineState: kline: got 0x00
2024-03-31_08_58_31_090: EngineState: kline: got 0x1F
2024-03-31_08_58_31_090: EngineState: kline: got 0x00
2024-03-31_08_58_31_090: EngineState: kline: got 0x00
2024-03-31_08_58_31_090: EngineState: kline: got 0x44
2024-03-31_08_58_31_090: EngineState: kline: got count 0x05
2024-03-31_08_58_31_090: EngineState: kline: got 0x00
2024-03-31_08_58_31_091: EngineState: kline: got 0xD0
2024-03-31_08_58_31_091: EngineState: kline: got 0x04
2024-03-31_08_58_31_091: EngineState: kline: got 0x00
2024-03-31_08_58_31_091: EngineState: kline: got 0x35
2024-03-31_08_58_31_091: EngineState: happy CRC 0x35
2024-03-31_08_58_31_091: EngineState: honda status packet with 0xD0 0x04 state 0
2024-03-31_08_58_31_091: EngineState: kline doSend
2024-03-31_08_58_31_092: EngineState: kline: got count 0x01
2024-03-31_08_58_31_092: EngineState: kline: got 0x02
2024-03-31_08_58_31_092: EngineState: kline: got count 0x03
2024-03-31_08_58_31_092: EngineState: kline: got 0x04
2024-03-31_08_58_31_092: EngineState: kline: got 0x00
2024-03-31_08_58_31_092: EngineState: kline: got 0x44
`

rusefillc, 2024-05-26:
@ElDominio OMG I've forgotten about one little detail :(

* please update firmware
* in the new menu please set "50"
* give it a try

[image]

closed 2024-06-14 by rusefillc

rusefillc, 2024-06-15:
@eldominio works with magic setting "100"

[image]

ElDominio, 2024-09-06:
Confirmed working!