rusefi/hellen125honda-issues #46: External aem wideband does not work over CAN

ElDominio, 2024-10-22:
I set the onboard wideband to index 1, and after reboot, it seems to set itself to index 0, showing overlapping wideband statuses in the gauge[image]

rusefillc, 2024-10-23:
See also https://github.com/rusefi/rusefi/issues/6980

ElDominio, 2024-10-26:
This is happening to a Hyundai154 too. It was not happening, I updated the firmware and now it's happening. Is newer firmware telling all controllers to get set to index 0?

rusefillc, 2024-10-26:
@ElDominio really doubt anything was changed :(

ElDominio, 2024-10-31:
https://youtu.be/s4iHTNE2S2A

You can see it on h254, this car has had the index of the onboard wb changed multiple times and it changes back after complete power cycle

rusefillc, 2024-10-31:
[image]

[image]

@ElDominio please try sending 01 or maybe even 02 to 

decimal 250871808 hex EF40000 (EXT obviously)

if that would work I would add a checkbox to do that with native code?

ElDominio, 2024-11-01:
txCan(1, 0xEF40000, 1, {01})
	
	Sending either 01 or 02 fixes it UNTIL NEXT POWER CYCLE. When the ECU is power cycled, it reverts to the conflicting reading between external CAN and AEM